gerritv/Grbl-Panel #119: default file extension not working

ljbuller, 2017-07-26:
grbl-panel 1.0.9.15

the default file extension text box does not populate from Grbl-panel settings DefaultFileExt.
I have txt in that field and the text box remains blank.

with the gui showing, you can type in nc and it will save it, but you can't type in txt.  it will accept the t, but ignores the x.  

I can't follow the code to see where it is saving this, but when typed in the gui, it is not saving it to visual studio project - grbl-panel properties - settings.

gerritv, 2017-08-01:
I have a fix, will push a release once I get a few more things off the list.

ljbuller, 2017-08-01:
Thanks Gerrit,

Also thank you for your great work in GrblPanel.  I have done quite a lot of work in VB.Net, and I am learning a lot on the threading and partial classes.

Larry

gerritv, 2017-08-01:
I'm not sure the partial classes were a good idea or not :-) My plan is still to rewrite in C# with better architecture, just don't know when that might happen. The implementation got a bit away from me and there are interconnections between the 'classes' that are a bit suspect.
I need to get a CNC machine working again before I proceed :-)

ljbuller, 2017-08-02:
Hi Gerrit,

What would c# give you that vb does not since it all compiles to the same code?

 

It seems like since  the receive event has to cycle through all the partial classes, it could just as easily do a few if/thens and branch out to the appropriate subs since they are all fairly short.

 

I see that it generates a thread for each partial class.  Have you studied the timing on those? 

 

Where do you live?  I am in Bismarck ND, USA.

 

I build the Panel Pro www.bullerent.com  buy one of those.  It is geared towards the aviation community making aluminum sheet metal products like instrument panels.

 

I have been using the Deskwin controller chip under license, but the author won’t update it so we can use com ports > 8.  Also it is on an obsolete chip that there is no guarantee of future production.  I won’t release the source code.  So far the Grbl looks quite promising.  I was all enthused about the udoo x86.  It runs win10 very nicely and has a built in Arduino 101 based on the Intel curie.  However Intel just announced they are dropping the curie. L

Larry

gerritv, 2017-08-02:
>What would c# give you that vb does not since it all compiles to the same code?

C# is the language I am using most now, just want to reduce the # of languages I work with. Also more code examples for MVVM patterns etc. I agree that it makes no difference to the runtime efficiency, one of the beauties of .Net.

>It seems like since  the receive event has to cycle through all the partial classes, it could just as easily do a few if/thens and branch out to the appropriate subs since they are all fairly short.    I see that it generates a thread for each partial class.  Have you studied the timing on those?

The partial classes that are called are actually full classes/objects. It is the main gui class/thread that is the partial class in each object file. E.g. Position is a class that processes the Mpos etc parts of the Grbl response to '?', the partial GUI class is in that file to keep the related gui elements in context. Using Delegates/Invoke for receive data lets me keep a responsive GUI while getting the other work done in background. It is essentially a callback list. It also lets me add objects to do other work, basically plug them into the Grbl response bus using Add Delegate for the receive event without physically adding calls for subroutines.
I did quite a bit of performance tuning while getting the load time of hug files down to acceptable level. While there might be a bunch of threads active, the bottlenecks were elsewhere.
For my convenience/laziness I do too many changes at gui level from other places, not a very clean architecture. E.g. using textbox.text as a data container that is retrieved elsewhere is not very good design :-) There are a few enhancements on hold because of the untidiness. A rewrite would make for a more robust design for longer term. 

> Where do you live?  I am in Bismarck ND, USA. 

St Catharines, Ontario, in the best wine country of North America, 60+ wineries, fruit farms, flowers etc. close to Niagara Falls and far enough away from Toronto that I don't miss that place :-)

> I build the Panel Pro www.bullerent.com  buy one of those.  It is geared towards the aviation community making aluminum sheet metal products like instrument panels.    I have been using the Deskwin controller chip under license, but the author wonâ€™t update it so we can use com ports > 8.  Also it is on an obsolete chip that there is no guarantee of future production.  I wonâ€™t release the source code.  

My budget is limited, retirement has its plusses and minuses :-) Looks like a good machine.

>So far the Grbl looks quite promising.  I was all enthused about the udoo x86.  It runs win10 very nicely and has a built in Arduino 101 based on the Intel curie.  However Intel just announced they are dropping the curie. 

The ARM version will open up a lot more code space to add missing features such as tool tables, canned cycles etc. Sadly hw obsolescence is a fact of life. Choosing a standard form factor should provide some isolation from specific vendor decisions, as long as there is Ethernet and usb connections. Serial is of course easier to provide electrical isolation for so a good thing to keep on board.

--

gerritv, 2017-11-09:
Fixed the original portion of this issue.

closed 2017-11-09 by gerritv