gerritv/Grbl-Panel #120: Found Override Feed has some problems?

viewsat, 2017-07-29:
When the send G-code, the Override Feed down to 10%, and then adjust the Feed up 100% did not respond, it seems that there are some problems.
I have also tested other Gcode sender programs to work properly.

I use the G-code:
G21 
F100
G1X200
G1Y150
G1Z50
G1X0
G1Y0
G1Z0
G1X200
G1Y150
G1Z50
G1X0
G1Y0
G1Z0
M30

gerritv, 2017-07-29:
I don't have a physical machine at present but the Status reports from Grbl definitely show the FS part of the response changing as Feed Override is changed up and down.
I tested with Grbl 1.1e so not the latest

viewsat, 2017-07-30:
The following two pictures of my Grbl settings, when x or y or z [accel] is set to 150 ($ 120 or $ 121 or 122 = 150). Send gcode F = 100, Override Feed down to 10, There will be an error.
I use Grbl-Panel to test Grbl 1.1e & 1.1f results all have the same error.
I use candle 1.1.7 gcode sender test no error occurred.

[image]
[image]

viewsat, 2017-07-31:
@gerritv Please watch my recorded video to understand this error.

see vedio>> https://youtu.be/qFnzyvqSxfk

gerritv, 2017-07-31:
https://github.com/gnea/grbl/issues/238#issuecomment-319182351

closed 2017-07-31 by gerritv