gerritv/Grbl-Panel #121: NullReferenceException was unhandled

trietdao95, 2017-08-18:
I want to compile the source code myself but there is a problem in GrblJogging.vb at line 24:
_gui.btnXPlus.Interval = 1000 / .JoggingXRepeat
It said NullReferenceException was unhandled. I'm using Visual Studio 2015. Hope you can help me with this issue
[image]

gerritv, 2017-08-18:
I used to use 2105, now 2017. I stepped through the code just now and don't get an exception. 

After you Stop debugging, is there a value/setting in Project/Settings for .JoggingXRepeat? Or is it btnXPlus that is Null?

trietdao95, 2017-08-20:
I redownload the code and it works fine now. Thank you!

gerritv, 2017-08-20:
Great news!

closed 2017-08-20 by gerritv