gerritv/Grbl-Panel #125: Feed Rates are base on?

tiptonspiderj, 2017-09-27:
gerritv,

I am new to using Grbl Panel and I can't find any documentation regarding the feed rate numbers next to the radio buttons.  Specifically I am looking for how you came up with the numbers for Metric...1000, 500, 100, and 5.  Any information would be much appreciated.

Sincerely,
Jeremy

gerritv, 2017-09-27:
They are not unusual rates for jogging. Perhaps a misleading name, maybe should have used Jog Rates. The values can be altered to suit a particular situation on the Settings page.

tiptonspiderj, 2017-09-28:
It's not the rates that confuse me, but why there is no correlation between the Metric values and the Imperial values for all four values.  I am a programmer looking at your code and there is no explanation of where the numbers come from.  What does the feed rate of 1000 stand for in Metric?  I understand you can change the reference in the settings page...I don't understand why the first three numbers correlate mathematically and the number 5 doesn't correspond.

tiptonspiderj, 2017-09-28:
I am looking for how the Metric numbers correlate to the Imperial numbers for your feed rate calculations.

gerritv, 2017-09-28:
They are just numbers. no correlation intended or expected between inferial and metric. The values are only used for jogging, so I choose values that seemed relevant to each measurement system.

closed 2017-09-28 by gerritv

tiptonspiderj, 2017-09-28:
Your response makes no sense at all.  How can you say something should equal a value when you have no reference for it?

tiptonspiderj, 2017-09-28:
I guess this goes towards your whole approach for programming with partial classes in visual basic.

gerritv, 2017-09-28:
I'm not following your path in why the question relates to code. Feed rates are rates related to machine movement, chosen to accomplish some movement to jog to a location. Nothing to do with programming style.

tiptonspiderj, 2017-09-28:
How can you say a default should be XXXX number when you don't know where the number comes from?

tiptonspiderj, 2017-09-28:
You have already answered my question in the fact that you simply do not know.

Sunnysky, 2017-09-28:
Feedrates are defined in the program and saved on exit. 
I used 60000 mm/min to get 1 m/s velocity 
What is it you need to understand?

From: tiptonspiderj 
Sent: Wednesday, September 27, 2017 10:53 PM
To: gerritv/Grbl-Panel 
Cc: Subscribed 
Subject: Re: [gerritv/Grbl-Panel] Feed Rates are base on? (#125)

How can you say a default should be XXXX number when you don't know where the number comes from?