gerritv/Grbl-Panel #127: Feature Request: Macro Buttons

FKjolner, 2017-11-06:
The possibility to use Macro Buttons from the "front page" 

I know these are supposed to be mapped to F2-Fsomething, but I'm really not in "keyboard mode" when doing my milling.

Further suggestions on the topic:
1. Place the buttons directly under the manual command text field. 
2. only show buttons for macros that exist, OR grey the non-existent out.
2. Button text option in the macro editor. as to define the text on the button (with a limited number of characters. )
3. allow for 4-5 macros, leave the rest for the F-keys.

This is the ONLY thing I'm missing from GRBL panel at the moment, GREAT product!!

Thanks&
Best Regards
Fredrik Kjølner.

joebananas10, 2017-11-06:
Don't items one through three already exist? My screen already looks like that. Am I missing something? 
[image]
Does your interface look different?

FKjolner, 2017-11-08:
I think there is something wrong with how the settings saves and updates. I see what you see now, but there are strange things in how it updates. I need to do more tests. And probably close the Feature Request, open a bug or improvement instead.

Thanks&
Best Regards
Fredrik Kjølner

gerritv, 2017-11-09:
There is a persistent problem with settings/macros not propagating when you use a newer version of GrblPanel. I have tried a few things to get control over that but so far no luck in solving it consistently.
The Settings are kept in C:\Users\<username>\AppData\Local\GrblPanel in subdirectories. I would look at the dates and copy 'user.config' from previous dated directory to newest directory.

closed 2017-11-09 by gerritv