gerritv/Grbl-Panel #128: Three digit T codes causes unhandled exception

axcodeplex, 2017-11-08:
T codes using numbers larger than 99 cause a program fault.
Examples:
T88 - result OK
T300 - unhandled exception

Steps to repro: Send a three digit T code via mdi or file read.

************** Exception Text **************
System.Collections.Generic.KeyNotFoundException: The given key was not present in the dictionary.
   at System.ThrowHelper.ThrowKeyNotFoundException()
   at System.Collections.Generic.Dictionary`2.get_Item(TKey key)
   at GrblPanel.GrblGui.showGrblStatus(String data)
   at GrblPanel.GrblIF.raiseAppSerialDataEvent(String data)
   at GrblPanel.GrblIF.VB$StateMachine_40__client_ComReadData.MoveNext()
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
----------------------------------------"

gerritv, 2017-11-08:
I will add Error 38 text but Grbl will still spit back:
"38","Invalid gcode ID:38","Tool number greater than max supported value."

gerritv, 2017-11-09:
Fixed missing error message

closed 2017-11-09 by gerritv