gerritv/Grbl-Panel #129: No issue, Video

axcodeplex, 2017-11-12:
I put together a walkthrough of this Excellent tool running a Shapeoko 3:

https://www.youtube.com/watch?v=kGbxkIHce9U
Let me know if anything in there is hilariously wrong  and I'll try and get it corrected.

Thanks @gerritv for the fantastic work on grbl-panel!!

gerritv, 2017-11-12:
Well done!
A question, is it common on your machine that you have to do the Settings refresh? On some Grbl boards a longer start up delay is required, thus the parameter on the settings machine. If using Bluetooth there is a similar symptom due to the DTR signal not being replicated at the Grbl end (used to reset Grbl itself).

axcodeplex, 2017-11-12:
You know, I never even questioned the use of the "Startup Delay" field. :-( In short, yes, have to do a manual refresh on every connect, but now that I know about that field I'll have to give it a crack.

aeleus, 2017-11-20:
Thanks axcodeplex! I learned how to use work positions and soft limits with your video and now don't worry about crashes.